myny-git/smappee_ev #14: [BUG]

ballydoon, 2025-08-13:
## 📢 Before You Begin

❗ **Issues are *not* for general questions or how-tos.**  
👉 Please use the [Discussions section](https://github.com/myny-git/smappee_ev/discussions) to ask questions or share ideas.

## ⚠️ Check for the Latest Version in HACS

HACS does not always notify you of updates automatically. Please:

1. Go to `HACS` > Integrations in Home Assistant  
2. Search for `smappee`  
3. Click the 3-dot menu next to the integration  
4. Choose `Update Information`  
5. If an update is available, install it **before reporting this issue**

## ✅ Pre-Check Checklist

Please confirm **all** of the following before submitting:

- [ ] I confirm that no other conflicting Smappee EV integrations are installed or configured (including deactivated ones — the official Smappee integration does not count).
- [ ] My Home Assistant version is up to date.
- [ ] I am using the **latest version** of this integration (verified via HACS).
- [ ] I have reviewed [open and closed issues](https://github.com/myny-git/smappee_ev/issues?q=is%3Aissue) to avoid duplicates.
- [ ] I have enabled and prepared DEBUG log output (for bug reports).
- [ ] This is a bug report — not a general usage question.

## 📋 What are the steps to reproduce this issue?

1. …
2. …
3. …

## ❗ What happens?

...

## ✅ What were you expecting to happen?

...

## 🪵 Any logs, error output, etc?

_Please paste any relevant **DEBUG log output** below:_

## 📎 Any other comments (or screenshots)?

...

myny-git, 2025-08-13:
Hi ballydoon, you created a bug, but did not put any messages or text ... can you add this, then, I can try to solve it.
Also indicate the version you use

ballydoon, 2025-08-13:
[image]
[image]

See attached images. I set the 28A max on the Smappee dashboard over a year ago.

myny-git, 2025-08-13:
Ok, thanks! So, then, it would be nice to do a quick curl:

curl -X GET https://app1pub.smappee.net/dev/v3/servicelocation/SERVICELOCATIONID/smartdevices/SMARTDEVICEID
-H "Authorization: Bearer YOURBEARER"  
-H "Content-Type: application/json"  

```json
    "configurationProperties": [{
        "value": {
            **"value": 32,**
            "unit": "A"
        },
        "spec": {
            **"name": "etc.smart.device.type.car.charger.config.max.current",**
            "displayName": "Maximum current",
            "description": "The maximum current that this charging station can support.",
            "species": "Quantity",
            "unit": "A",
            "required": true,
            "possibleValues": {
                "values": [],
                "exhaustive": true
            },
```
if there the value is present, then I have to somehow update the dynamic usage of this number.

ballydoon, 2025-08-13:
Sorry - not sure how to do that

ballydoon, 2025-08-13:
Great - thanks

ballydoon, 2025-08-13:
Looks to be working fine.